google-deepmind/mujoco_warp #17: The default `mjwarp-testspeed` fails to run

hartikainen, 2025-03-21:
System information:

```shell
$ pwd
/home/user/google-deepmind/mujoco_warp
$ git rev-parse HEAD
49c9ecd40c32d9706d84725d4a3b2e8e056ff39d
$ nvidia-smi --query-gpu=name --format=csv,noheader
NVIDIA L4
$ python --version
Python 3.12.3
$ uname -a
Linux xxx 6.8.0-1021-gcp #23-Ubuntu SMP Wed Jan 15 22:37:41 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
```

I'm trying to run the default testpeed example from the main README:
```shell
mjwarp-testspeed \
  --function=step \
  --base_path=./mujoco_warp/test_data/humanoid \
  --mjcf=humanoid.xml \
  --clear_kernel_cache=true
```

This results in segmentation fault, however. I've tried varying all the parameters of the script, e.g. in the hope that decreasing the model size would help, but they make no difference.

<details>

<summary>Full error</summary>

```
$ mjwarp-testspeed \
  --function=step \
  --base_path=./mujoco_warp/test_data/humanoid \
  --mjcf=humanoid.xml \
  --clear_kernel_cache=true
Warp 1.7.0.dev20250321 initialized:
   Git commit: f3814e7e5459e5fd13032cf0fddb3daddd510f30
   CUDA Toolkit 12.8, Driver 12.4
   Devices:
     "cpu"      : "x86_64"
     "cuda:0"   : "NVIDIA L4" (22 GiB, sm_89, mempool enabled)
   Kernel cache:
     /home/user/.cache/warp/1.7.0.dev20250321
Model nbody: 17 nv: 27 ngeom: 20 is_sparse: False solver: newton
Params nconmax: -1 njmax: -1
Data ncon: 8 nefc: 32 keyframe: 0
Linesearch: iterative
Rolling out 1000 steps at dt = 0.005...
Module mujoco_warp._src.smooth.kinematics 2b57258 load on device 'cuda:0' took 2582.64 ms  (compiled)
Module mujoco_warp._src.smooth.com_pos f3e2e97 load on device 'cuda:0' took 1463.57 ms  (compiled)
Module mujoco_warp._src.warp_util 6e198ea load on device 'cuda:0' took 790.14 ms  (compiled)
Module mujoco_warp._src.smooth.crb 0b76250 load on device 'cuda:0' took 915.43 ms  (compiled)
Module mujoco_warp._src.smooth._factor_i_dense.tile_cholesky 55c393e load on device 'cuda:0' took 435.34 ms  (compiled)
Module mujoco_warp._src.collision_driver edb1808 load on device 'cuda:0' took 2001.75 ms  (compiled)
Module mujoco_warp._src.collision_primitive 022b717 load on device 'cuda:0' took 3198.33 ms  (compiled)
Module mujoco_warp._src.constraint c0fe614 load on device 'cuda:0' took 2661.22 ms  (compiled)
Module mujoco_warp._src.smooth.transmission 1873113 load on device 'cuda:0' took 853.33 ms  (compiled)
Module mujoco_warp._src.forward.fwd_velocity.actuator_velocity 43c9064 load on device 'cuda:0' took 1016.05 ms  (compiled)
Module mujoco_warp._src.smooth.com_vel cad50c8 load on device 'cuda:0' took 2020.46 ms  (compiled)
Module mujoco_warp._src.passive.passive f34c5b3 load on device 'cuda:0' took 1059.52 ms  (compiled)
Module mujoco_warp._src.smooth.rne 7879965 load on device 'cuda:0' took 1129.21 ms  (compiled)
Module mujoco_warp._src.forward.fwd_actuation 65cf35c load on device 'cuda:0' took 590.91 ms  (compiled)
Module mujoco_warp._src.forward.fwd_actuation.qfrc_actuator 0824cb9 load on device 'cuda:0' took 1001.03 ms  (compiled)
Module mujoco_warp._src.forward.fwd_acceleration ed679f4 load on device 'cuda:0' took 383.05 ms  (compiled)
Module mujoco_warp._src.support 6d747d6 load on device 'cuda:0' took 554.78 ms  (compiled)
Module mujoco_warp._src.smooth._solve_LD_dense.tile_cho_solve 68c91bc load on device 'cuda:0' took 441.54 ms  (compiled)
Module mujoco_warp._src.solver._create_context 9e7d4cd load on device 'cuda:0' took 604.81 ms  (compiled)
Module mujoco_warp._src.support.mul_m.tile_mul 326c422 load on device 'cuda:0' took 1007.26 ms  (compiled)
Module mujoco_warp._src.solver._update_constraint 204c4d7 load on device 'cuda:0' took 861.67 ms  (compiled)
Module mujoco_warp._src.solver._update_gradient cd59223 load on device 'cuda:0' took 4012.05 ms  (compiled)
Module mujoco_warp._src.solver._update_gradient c46b2c6 load on device 'cuda:0' took 4660.84 ms  (compiled)
Module mujoco_warp._src.solver._linesearch 42342f2 load on device 'cuda:0' took 1057.40 ms  (compiled)
Module mujoco_warp._src.solver._linesearch_iterative 73e6649 load on device 'cuda:0' took 1641.79 ms  (compiled)
Module mujoco_warp._src.solver.solve 9f3bd70 load on device 'cuda:0' took 594.61 ms  (compiled)
Module mujoco_warp._src.forward._advance 9193f2d load on device 'cuda:0' took 1342.84 ms  (compiled)
Fatal Python error: Segmentation fault

Current thread 0x0000734b51684080 (most recent call first):
  File "/home/user/google-deepmind/mujoco_warp/.venv/lib/python3.12/site-packages/warp/context.py", line 5848 in launch
  File "/home/user/google-deepmind/mujoco_warp/mujoco_warp/_src/smooth.py", line 133 in kinematics
  File "/home/user/google-deepmind/mujoco_warp/mujoco_warp/_src/warp_util.py", line 70 in wrapper
  File "/home/user/google-deepmind/mujoco_warp/mujoco_warp/_src/forward.py", line 412 in fwd_position
  File "/home/user/google-deepmind/mujoco_warp/mujoco_warp/_src/warp_util.py", line 70 in wrapper
  File "/home/user/google-deepmind/mujoco_warp/mujoco_warp/_src/forward.py", line 668 in forward
  File "/home/user/google-deepmind/mujoco_warp/mujoco_warp/_src/warp_util.py", line 70 in wrapper
  File "/home/user/google-deepmind/mujoco_warp/mujoco_warp/_src/forward.py", line 685 in step
  File "/home/user/google-deepmind/mujoco_warp/mujoco_warp/_src/warp_util.py", line 70 in wrapper
  File "/home/user/google-deepmind/mujoco_warp/mujoco_warp/_src/test_util.py", line 102 in benchmark
  File "/home/user/google-deepmind/mujoco_warp/mujoco_warp/testspeed.py", line 105 in _main
  File "/home/user/google-deepmind/mujoco_warp/.venv/lib/python3.12/site-packages/absl/app.py", line 261 in _run_main
  File "/home/user/google-deepmind/mujoco_warp/.venv/lib/python3.12/site-packages/absl/app.py", line 316 in run
  File "/home/user/google-deepmind/mujoco_warp/mujoco_warp/testspeed.py", line 187 in main
  File "/home/user/google-deepmind/mujoco_warp/.venv/bin/mjwarp-testspeed", line 10 in <module>

Extension modules: numpy._core._multiarray_umath, numpy.linalg._umath_linalg (total: 2)
[1]    1811387 segmentation fault (core dumped)  mjwarp-testspeed --function=step --base_path=./mujoco_warp/test_data/humanoid
```
</details/

Swetanshu575, 2025-03-21:
A segmentation fault when running the mjwarp-testspeed example from the mujoco_warp README suggests a low-level issue, likely related to memory access, library compatibility, or environment configuration. Since you’ve already tried varying parameters (e.g., model size) without success, the problem is likely not tied to the specific MJCF file or model complexity but rather to the setup or execution environment. Let’s diagnose and resolve this step-by-step.

erikfrey, 2025-03-22:
@hartikainen Thanks for the bug report.  It's early days and we don't yet have a very good understanding of how we perform across various GPUs and CUDA versions.  Fwiw, I also had some issues until I updated both CUDA (12.8) and NVIDIA drivers (570.124.06).

After the rush of feature development settles, we'll start to analyze and document more carefully what versions of CUDA we support, and also fix any low hanging fruit to be as backwards compatible as we reasonably can.

In the meantime if you'd like to continue to dig into this, one good first step is install [NVIDIA Warp](https://github.com/NVIDIA/warp) separately and make sure all the tests pass there and you see no segfaults.  Once that's established we have a good datapoint that we're doing something wrong on our end and can dig in a bit further.

I'll leave this open for now in case you'd like to keep poking, otherwise we'll come back to it in a month or so.

hartikainen, 2025-03-22:
I installed warp separately as per the README instructions. None of the tests segfault, although I do see a couple of test failures.

<details>

<summary>Test output</summary>

```
$ pytest
================================ test session starts ================================
platform linux -- Python 3.12.3, pytest-8.3.5, pluggy-1.5.0
rootdir: /home/user/google-deepmind/mujoco_warp
configfile: pyproject.toml
collected 52 items

mujoco_warp/_src/broad_phase_test.py ..                                       [  3%]
mujoco_warp/_src/collision_driver_test.py .........                           [ 21%]
mujoco_warp/_src/constraint_test.py .                                         [ 23%]
mujoco_warp/_src/forward_test.py .........                                    [ 40%]
mujoco_warp/_src/io_test.py ...                                               [ 46%]
mujoco_warp/_src/math_test.py .......                                         [ 59%]
mujoco_warp/_src/passive_test.py .                                            [ 61%]
mujoco_warp/_src/smooth_test.py ....F...F..                                   [ 82%]
mujoco_warp/_src/solver_test.py .....                                         [ 92%]
mujoco_warp/_src/support_test.py ....                                         [100%]

===================================== FAILURES ======================================
_____________________________ SmoothTest.test_factor_m0 _____________________________

self = <mujoco_warp._src.smooth_test.SmoothTest testMethod=test_factor_m0>
sparse = True

    @parameterized.parameters(True, False)
    def test_factor_m(self, sparse: bool):
      """Tests factor_m."""
      _, mjd, m, d = test_util.fixture("pendula.xml", sparse=sparse)

      qLD = d.qLD.numpy()[0].copy()
      for arr in (d.qLD, d.qLDiagInv):
        arr.zero_()

      mjwarp.factor_m(m, d)

      if sparse:
>       _assert_eq(d.qLD.numpy()[0, 0], mjd.qLD, "qLD (sparse)")

mujoco_warp/_src/smooth_test.py:104:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

a = array([ 7.8662987e+00,  1.3947523e+01,  1.9630328e-01,  3.4761963e+01,
        4.5368922e-01,  2.3626085e-01,  6.33546...  5.9126085e-01, -1.6744296e-01, -1.0686923e+00,  1.0764145e+00,
       -2.1517043e+00,  1.0762539e+00], dtype=float32)
b = array([ 7.86629749e+00,  1.96303266e-01,  1.39475238e+01,  2.36260852e-01,
        4.53689186e-01,  3.47619650e+01, -1...0,
        1.07641440e+00, -1.06869235e+00, -1.67442963e-01,  5.91260866e-01,
       -4.95247904e-05,  1.92164948e+00])
name = 'qLD (sparse)'

    def _assert_eq(a, b, name):
      tol = _TOLERANCE * 10  # avoid test noise
      err_msg = f"mismatch: {name}"
>     np.testing.assert_allclose(a, b, err_msg=err_msg, atol=tol, rtol=tol)
E     AssertionError:
E     Not equal to tolerance rtol=0.0005, atol=0.0005
E     mismatch: qLD (sparse)
E     Mismatched elements: 80 / 98 (81.6%)
E     Max absolute difference among violations: 95.56416283
E     Max relative difference among violations: 3.74131147e+18
E      ACTUAL: array([ 7.866299e+00,  1.394752e+01,  1.963033e-01,  3.476196e+01,
E             4.536892e-01,  2.362608e-01,  6.335462e+00,  1.065826e+00,
E            -1.528473e+00, -1.428232e-01,  3.808000e+00, -5.924370e-01,...
[... 45 lines cut ...]
      err_msg = f"mismatch: {name}"
>     np.testing.assert_allclose(a, b, err_msg=err_msg, atol=tol, rtol=tol)
E     AssertionError:
E     Not equal to tolerance rtol=0.0005, atol=0.0005
E     mismatch: qacc_smooth
E     Mismatched elements: 31 / 33 (93.9%)
E     Max absolute difference among violations: 662355.85454353
E     Max relative difference among violations: 5.69995407e+09
E      ACTUAL: array([ 2.069320e+03, -2.882091e+04, -5.886675e+04, -7.236778e+04,
E            -1.703670e+04, -3.350715e+04, -4.446414e+01,  1.821589e+02,
E             3.060436e+02, -7.863045e+00, -1.975973e-04,  1.379908e+04,...
E      DESIRED: array([ 1.039810e-05,  1.013633e-05, -9.809994e+00,  1.269620e-05,
E             2.154922e-05, -9.681395e-06, -1.084511e+01,  6.414392e+00,
E             7.080158e-01, -7.863045e+00, -1.975973e-04, -6.272711e-02,...

mujoco_warp/_src/smooth_test.py:38: AssertionError
============================== short test summary info ==============================
FAILED mujoco_warp/_src/smooth_test.py::SmoothTest::test_factor_m0 - AssertionError:
FAILED mujoco_warp/_src/smooth_test.py::SmoothTest::test_solve_m0 - AssertionError:
====================== 2 failed, 50 passed in 99.54s (0:01:39) ======================
```

</details>

Happy to provide more diagnostics if you get to debugging this at any point and need more information.

saran-t, 2025-03-22:
This is the same failure I see internally. What MuJoCo version are you running?

hartikainen, 2025-03-23:
I was using `mujoco==3.3.0`. Downgrading to `mujoco==3.2.7` removes the test errors. The original segfault remains the same, though.

<details>

<summary>Test output with `mujoco==3.2.7`</summary>

```shell
$ pytest
================================ test session starts ================================
platform linux -- Python 3.12.3, pytest-8.3.5, pluggy-1.5.0
rootdir: /home/user/google-deepmind/mujoco_warp
configfile: pyproject.toml
collected 52 items

mujoco_warp/_src/broad_phase_test.py ..                                       [  3%]
mujoco_warp/_src/collision_driver_test.py .........                           [ 21%]
mujoco_warp/_src/constraint_test.py .                                         [ 23%]
mujoco_warp/_src/forward_test.py .........                                    [ 40%]
mujoco_warp/_src/io_test.py ...                                               [ 46%]
mujoco_warp/_src/math_test.py .......                                         [ 59%]
mujoco_warp/_src/passive_test.py .                                            [ 61%]
mujoco_warp/_src/smooth_test.py ...........                                   [ 82%]
mujoco_warp/_src/solver_test.py .....                                         [ 92%]
mujoco_warp/_src/support_test.py ....                                         [100%]
```

</details>

arpit15, 2025-03-27:
Adding more debug info. I added to the top of viewer script `wp.config.print_launches = True` to get more debug info. The issue seems to be in codegen due to cudagraph capture. 
Redacted log is shown below
<details>
Warp CUDA error 906: operation would make the legacy stream depend on a capturing blocking stream (in function cuda_context_set_stream, /builds/omniverse/warp/warp/native/warp.cu:2151)
Warp CUDA error 906: operation would make the legacy stream depend on a capturing blocking stream (in function memcpy_d2h, /builds/omniverse/warp/warp/native/warp.cu:754)
Traceback (most recent call last):
  File "/home/user/projects/mujoco_warp/mujoco_warp/viewer.py", line 92, in _main
    mjwarp.step(m, d)
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/warp_util.py", line 101, in wrapper
    return fn(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/forward.py", line 684, in step
    forward(m, d)
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/warp_util.py", line 101, in wrapper
    return fn(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/forward.py", line 667, in forward
    fwd_position(m, d)
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/warp_util.py", line 101, in wrapper
    return fn(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/forward.py", line 412, in fwd_position
    smooth.kinematics(m, d)
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/warp_util.py", line 101, in wrapper
    return fn(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/smooth.py", line 133, in kinematics
    wp.launch(_root, dim=(d.nworld), inputs=[m, d])
  File "/home/user/projects/mujoco_warp/.venv/lib/python3.12/site-packages/warp/context.py", line 5689, in launch
    print(f"kernel: {kernel.key} dim: {dim} inputs: {inputs} outputs: {outputs} device: {device}")
                                                    ^^^^^^^^
  File "/home/user/projects/mujoco_warp/.venv/lib/python3.12/site-packages/warp/codegen.py", line 344, in __repr__
    return struct_instance_repr_recursive(self, 0)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/.venv/lib/python3.12/site-packages/warp/codegen.py", line 221, in struct_instance_repr_recursive
    lines.append(f"{indent * (depth + 1)}{field_name}={field_value},")
                                                      ^^^^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/.venv/lib/python3.12/site-packages/warp/types.py", line 2236, in __str__
    return str(self.numpy())
               ^^^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/.venv/lib/python3.12/site-packages/warp/types.py", line 2574, in numpy
    a = self.to("cpu", requires_grad=False)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/.venv/lib/python3.12/site-packages/warp/types.py", line 2645, in to
    return warp.clone(self, device=device, requires_grad=requires_grad)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/.venv/lib/python3.12/site-packages/warp/context.py", line 5128, in clone
    warp.copy(arr, src)
  File "/home/user/projects/mujoco_warp/.venv/lib/python3.12/site-packages/warp/context.py", line 6382, in copy
    raise RuntimeError(f"Warp copy error: {runtime.get_error_string()}")
RuntimeError: Warp copy error: Warp CUDA error 906: operation would make the legacy stream depend on a capturing blocking stream (in function memcpy_d2h, /builds/omniverse/warp/warp/native/warp.cu:754)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/user/projects/mujoco_warp/mujoco_warp/viewer.py", line 135, in <module>
    main()
  File "/home/user/projects/mujoco_warp/mujoco_warp/viewer.py", line 131, in main
    app.run(_main)
  File "/home/user/projects/mujoco_warp/.venv/lib/python3.12/site-packages/absl/app.py", line 316, in run
    _run_main(main, args)
  File "/home/user/projects/mujoco_warp/.venv/lib/python3.12/site-packages/absl/app.py", line 261, in _run_main
    sys.exit(main(argv))
             ^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/mujoco_warp/viewer.py", line 91, in _main
    with wp.ScopedCapture() as capture:
         ^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/.venv/lib/python3.12/site-packages/warp/utils.py", line 955, in __exit__
    self.graph = wp.capture_end(device=self.device, stream=self.stream)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/.venv/lib/python3.12/site-packages/warp/context.py", line 6227, in capture_end
    raise RuntimeError(f"CUDA graph capture failed. {runtime.get_error_string()}")
RuntimeError: CUDA graph capture failed. Warp CUDA error 906: operation would make the legacy stream depend on a capturing blocking stream (in function memcpy_d2h, /builds/omniverse/warp/warp/native/warp.cu:754)
[ble: exit 1][ble: elapsed 60.404s (CPU 102.1%)] python mujoco_warp/viewer.py --mjcf=mujoco_warp/test_data/humanoid/humanoid.xml 

</details>
The warp concurrency [doc](https://nvidia.github.io/warp/modules/concurrency.html#cuda-default-stream) and relevant [issue](https://github.com/NVIDIA/warp/issues/545) clear state the issue. 

I am wondering if there is suggested fix.

thowell, 2025-07-03:
@hartikainen does `mjwarp-testspeed` work now?

hartikainen, 2025-07-03:
It does. I'm gonna close this. Thanks folks!

<details><summary>Details</summary>

```console
$ git rev-parse HEAD
12a370fb62e6ca67eb31b641455a19c6376e3edd
$ nvidia-smi --query-gpu=name --format=csv,noheader
NVIDIA L4
$ python --version
Python 3.12.3
$ python -c "import warp as wp; print(wp.__version__)"
1.9.0.dev20250703
$ uname -a
Linux workstation-kristian.us-central1-a.c.eka-robotics.internal 6.11.0-1015-gcp #15~24.04.1-Ubuntu SMP Thu Apr 24 20:41:05 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
```

</details>

closed 2025-07-03 by hartikainen