google-deepmind/mujoco_warp #7: Jupyter kernel dies at `train_fn` in `contrib/apptronik_apollo_locomotion.ipynb`

sachinpullil, 2025-03-21:
Hi,

My Jupyter kernel dies near the `train_fn` part of the given example notebook. Not much in terms of error messages:

```
21:11:08.263 [info] Kernel successfully started
21:11:08.283 [info] Process Execution: ~/external/mujoco_warp/mujoco_warp_venv/bin/python /home/~/.vscode/extensions/ms-toolsai.jupyter-2024.11.0-linux-x64/pythonFiles/printJupyterDataDir.py
21:11:37.767 [error] Disposing session as kernel process died ExitCode: undefined, Reason: 
21:12:19.996 [info] Restart requested ~/external/mujoco_warp/contrib/apptronik_apollo_locomotion.ipynb
21:12:19.998 [warn] Cancel all remaining cells due to dead kernel
```

I tried reducing the number of environments to 2048 (and NCONMAX to 20480). I also tried reducing the timesteps to `50_000`. No luck.

My `nvidia-smi` results if you need it:
```
nvidia-smi
Thu Mar 20 21:19:37 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.127.08             Driver Version: 550.127.08     CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3090        Off |   00000000:47:00.0  On |                  N/A |
|  0%   54C    P8             38W /  350W |    2259MiB /  24576MiB |     34%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      5053      G   /usr/lib/xorg/Xorg                            619MiB |
|    0   N/A  N/A      5158    C+G   ...libexec/gnome-remote-desktop-daemon        258MiB |
|    0   N/A  N/A      5197      G   /usr/bin/gnome-shell                           82MiB |
|    0   N/A  N/A      5847      G   ...erProcess --variations-seed-version         67MiB |
|    0   N/A  N/A      8072    C+G   warp-terminal                                  57MiB |
|    0   N/A  N/A     19860      G   ...07,262144 --variations-seed-version        160MiB |
|    0   N/A  N/A     20401      G   ...erProcess --variations-seed-version        284MiB |
|    0   N/A  N/A    447393      G   ...erProcess --variations-seed-version         26MiB |
|    0   N/A  N/A    717180      C   ...co_warp/mujoco_warp_venv/bin/python        638MiB |
+-----------------------------------------------------------------------------------------+
```

erikfrey, 2025-03-24:
Hi @sachinpullil - I suspect this is a dupe of #17 just showing up via a different expression.  Can you confirm that `mjwarp-testspeed` also segfaults for you?  If so we'll track this over there.

sachinpullil, 2025-03-31:
Yes, `mjwarp-testspeed` also segfaults.