google-deepmind/mujoco_warp #8: OSError access violation - on Windows 11

Zuwa92, 2025-03-21:
I get this issue when running the "viewer.py" script:

  File "C:\**\viewer.py", line 129, in <module>
    main()
  File "C:\**\viewer.py", line 125, in main
    app.run(_main)
  File "C:\**\app.py", line 316, in run
    _run_main(main, args)
  File "C:\**\app.py", line 261, in _run_main
    sys.exit(main(argv))
             ^^^^^^^^^^
  File "C:\**\viewer.py", line 85, in _main
    with wp.ScopedCapture() as capture:
         ^^^^^^^^^^^^^^^^^^
  File "C:\**\utils.py", line 955, in __exit__
    self.graph = wp.capture_end(device=self.device, stream=self.stream)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\**\context.py", line 6217, in capture_end
    result = runtime.core.cuda_graph_end_capture(device.context, stream.cuda_stream, ctypes.byref(graph_exec))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: exception: access violation reading 0x0000000000000290

NOTE:  I used C:\**\  just to redact file directories. GPU is RTX 3080; driver: 546.33. CUDA 12.3

Please assist.

Thank you.

erikfrey, 2025-03-22:
@Zuwa92  Thanks for the bug report, going to respond somewhat similarly as #17:

Right now I think most of us are primarily developing on linux.  We're also using relatively fresh CUDA install and drivers (for me: CUDA 12.8, driver 570.124.06).

I do see that [NVIDIA Warp](https://github.com/NVIDIA/warp) has Windows support, so maybe that's not related to the issue you're experiencing.

After the rush of feature development settles, we'll start to analyze and document more carefully what versions of CUDA we support, and also fix any low hanging fruit to be as backwards compatible as we reasonably can.

In the meantime if you'd like to continue to dig into this, one good first step is install [NVIDIA Warp](https://github.com/NVIDIA/warp) separately and make sure all the tests pass there and you see no segfaults.  Once that's established we have a good datapoint that we're doing something wrong on our end and can dig in a bit further.

I'll leave this open for now in case you'd like to keep poking, otherwise we'll come back to it in a month or so.

shi-eric, 2025-03-22:
Do the MJWarp unit tests all pass on your system? It also might help us to have a full error log starting with the Python command. As Eric said, this could be a Warp issue instead of one with MJWarp.

eric-heiden, 2025-03-22:
I encountered a similar issue before on Windows with graph capture and resolved it by updating the driver and CUDA toolkit to the latest version (I am running MjWarp on Windows 11 on driver 572.61 with CUDA 12.8).

Zuwa92, 2025-03-22:
Thanks for the replies! Interesting, the replies from so many eric(k)s lol.
@eric-heiden I may try updating but I'm a bit hesitant because of other program driver/CUDA dependencies.
@shi-eric  I have attached the full log from running the viewer script - note that the "[REDACTED]" lines just replace my root directory. 
From inspection, it does seem to be more related to the Warp library, perhaps an issue with the latest nightly build? I have been using Warp for GPU-based simulation/computation for about a year now but haven't had a similar issue as far as I recall.  

[Error_Log_redacted.txt](https://github.com/user-attachments/files/19404897/Error_Log_redacted.txt)

Zuwa92, 2025-03-25:
Hi All,

Thanks for the support again! The issue is now resolved.
@eric-heiden I updated my driver to 572.83 and my CUDA to 12.8; I am now able to launch a simulation and with the viewer.
However, for some mjcf files (like the n_humanoids.xml), the characters seem to have collisions completely disabled and just sink through the floor.

closed 2025-04-09 by Zuwa92