google-deepmind/mujoco_warp #23: Cartpole `step` results do not match `mj_step` on CPU

pculbertson-bdai, 2025-03-23:
Hi y'all, really excited about the project! I was playing around with it today and noticed that `step` appears to give strange results for a simple toy cartpole. I am probably totally doing something wrong, but wanted to flag it in case something bad is happening.

I'd also be happy to help sort this out if you have any hunches about what could be going wrong. 

Debug script:
```
"""
Test that mjw.step matches mujoco.mj_step
"""
import os
from pathlib import Path

import mujoco
import mujoco_warp as mjw
import numpy as np
import warp as wp

def test_mjw_step():
    """Test that mjw.step matches mujoco.mj_step."""
    wp.init()
    wp.set_device("cuda:0")
    
    # Load and initialize model
    path = Path("cartpole_no_sensors.xml")
    mjm = mujoco.MjModel.from_xml_path(str(path))
    mjd = mujoco.MjData(mjm)
    
    # Initialize simulation state with some random initial conditions
    mujoco.mj_resetDataKeyframe(mjm, mjd, 0)
    mjd.qpos[:] = np.random.uniform(-0.1, 0.1, mjm.nq)
    mjd.qvel[:] = np.random.uniform(-0.1, 0.1, mjm.nv)
    mujoco.mj_forward(mjm, mjd)
    
    # Create Warp model and data
    m = mjw.put_model(mjm)
    d = mjw.put_data(mjm, mjd, nworld=1, nconmax=512, njmax=512)
    
    # Generate random action for testing
    action = np.random.uniform(-1, 1, (mjm.nu,))
    
    # Save initial state for both rollouts
    initial_qpos = mjd.qpos.copy()
    initial_qvel = mjd.qvel.copy()
    
    # Run MuJoCo step
    mjd.ctrl[:] = action
    mujoco.mj_step(mjm, mjd)
    
    # Reset MuJoCo and Warp states to initial conditions
    mjd.qpos[:] = initial_qpos
    mjd.qvel[:] = initial_qvel
    mujoco.mj_forward(mjm, mjd)
    
    # Copy initial state to Warp
    qpos_warp = wp.array(initial_qpos.reshape(1, -1), dtype=wp.float32)
    qvel_warp = wp.array(initial_qvel.reshape(1, -1), dtype=wp.float32)
    wp.copy(d.qpos, qpos_warp)
    wp.copy(d.qvel, qvel_warp)
    
    # Apply action and step Warp
    action_warp = wp.array(action.reshape(1, -1), dtype=wp.float32)
    wp.copy(d.ctrl, action_warp)
    mjw.step(m, d)
    
    # Print debug information
    print("\nArray shapes:")
    print(f"MuJoCo qpos shape: {mjd.qpos.shape}")
    print(f"MuJoCo qvel shape: {mjd.qvel.shape}")
    print(f"Warp d.qpos shape: {d.qpos.shape}")
    print(f"Warp d.qvel shape: {d.qvel.shape}")
    
    print("\nControl values:")
    print(f"MuJoCo ctrl: {action}")
    print(f"Warp ctrl: {d.ctrl.numpy()[0]}")
    
    print("\nInitial states:")
    print(f"MuJoCo qpos: {initial_qpos}")
    print(f"MuJoCo qvel: {initial_qvel}")
    print(f"Warp qpos: {qpos_warp.numpy()[0]}")
    print(f"Warp qvel: {qvel_warp.numpy()[0]}")
    
    print("\nAfter step:")
    print(f"MuJoCo qpos: {mjd.qpos}")
    print(f"MuJoCo qvel: {mjd.qvel}")
    print(f"Warp qpos: {d.qpos.numpy()[0]}")
    print(f"Warp qvel: {d.qvel.numpy()[0]}")
    
    # Compare with high precision
    np.testing.assert_allclose(
        np.concatenate([mjd.qpos, mjd.qvel]),
        np.concatenate([d.qpos.numpy()[0], d.qvel.numpy()[0]]),
        rtol=1e-5,
        atol=1e-5,
        err_msg="Warp step does not match MuJoCo step"
    )
    
    print("\nTest passed!")

if __name__ == "__main__":
    test_mjw_step() 
```

Cartpole XML:
```
<mujoco model="cartpole">
<!-- Copyright (c) 2023 Boston Dynamics AI Institute LLC. All rights reserved. -->

  <asset>
    <texture name="blue_grid" type="2d" builtin="checker" rgb1=".02 .14 .44" rgb2=".27 .55 1" width="300" height="300" mark="edge" markrgb="1 1 1"/>
    <material name="blue_grid" texture="blue_grid" texrepeat="1 1" texuniform="true" reflectance=".2"/>
    <material name="beige" rgba=".7 .5 .3 1"/>
  </asset>

  <option timestep="0.04" integrator="implicitfast">
    <flag contact="disable"/>
  </option>

  <default>
    <default class="cart">
      <joint damping="10" type="slide" axis="1 0 0" range="-1.8 1.8" />
      <position kp="100" ctrlrange="-1.8 1.8" forcerange="-10 10"/>
    </default>
    <default class="pole">
      <joint damping="0" type="hinge" axis="0 1 0"/>
      <geom type="capsule" fromto="0 0 0 0 0 1" size="0.045" material="beige" mass=".1"/>
    </default>
  </default>

  <worldbody>
    <body name="cart" pos="0 0 0">
      <joint name="joint_cart" class="cart" />
      <geom name="cart" type="box" size="0.2 0.15 0.1" material="beige"  mass="1"/>
      <body name="pole" childclass="pole">
        <joint name="joint_pole" class="pole"/>
        <geom name="pole"/>
      </body>
    </body>
  </worldbody>

  <actuator>
    <position name="actuator_cart" joint="joint_cart" class="cart" />
    <!-- <motor name="actuator_cart" joint="joint_cart" ctrllimited="true" forcelimited="true" ctrlrange="-100 100" forcerange="-100 100" /> -->
  </actuator>

</mujoco> 
```

Relevant output:
```
Array shapes:
MuJoCo qpos shape: (2,)
MuJoCo qvel shape: (2,)
Warp d.qpos shape: (1, 2)
Warp d.qvel shape: (1, 2)

Control values:
MuJoCo ctrl: [0.45525741]
Warp ctrl: [0.45525742]

Initial states:
MuJoCo qpos: [0.09810708 0.0107087 ]
MuJoCo qvel: [0.01977989 0.02162478]
Warp qpos: [0.09810708 0.0107087 ]
Warp qvel: [0.01977989 0.02162478]

After step:
MuJoCo qpos: [0.09810708 0.0107087 ]
MuJoCo qvel: [0.01977989 0.02162478]
Warp qpos: [ 0.13870206 -0.04599188]
Warp qvel: [ 1.0148748 -1.4175144]
```

Relevant packages from `pip freeze`:
```
-e git+https://github.com/google-deepmind/mujoco_warp.git@370942d2a5e5639c71e39f927ff3aa0f1232f30b#egg=mujoco_warp
warp-lang==1.7.0.dev20250322
```

pculbertson-bdai, 2025-03-23:
Found the culprit - the `forcerange` value on the actuator is causing problems. Reading the code [here](https://github.com/google-deepmind/mujoco_warp/blob/370942d2a5e5639c71e39f927ff3aa0f1232f30b/mujoco_warp/_src/forward.py#L540) I think the range is getting read from the `MjModel` but there's no clamping happening (vs. how the ctrlrange a couple lines above).

I can put up a one-line PR to add the clamp if this is a bugfix!

erikfrey, 2025-03-23:
Hi @pculbertson-bdai - whoops, you're right, we lost a line somewhere, thanks for spotting.  A PR would be great!

PR #24 (2025-03-23): Clamp actuator force to lie in force range specified in XML
Adds a clamp to the `forward` method to enforce the `forcerange` specified in the XML. Fixes #23. 

All unit tests passed after the change. Not sure if you'd like me to add tests to catch this moving forward.
closed 2025-03-23 by erikfrey