google-deepmind/mujoco_warp #25: Any plan to support the friction cone type :elliptic

louhz, 2025-03-23:
In many MJCF configurations for robotic hands and arms, elliptical friction is commonly used.
I'm curious about the potential challenges involved in adding support for this feature in mujoco_warp.
Would you consider this a technically difficult task, or is it simply a lower-priority item on the roadmap?

louhz, 2025-03-23:
Looks like main barrier to supporting elliptical friction is the lack of a Projected Gauss-Seidel (PGS) solver. So, would it be correct to say that we first need to implement the Gauss-Seidel solver before elliptical friction can be supported?

erikfrey, 2025-03-24:
Hello @louhz !  Yes indeed we will support elliptic cone.  Actually it does not require PGS.

We're still thinking about how to communicate our upcoming development roadmap, but I expect that soon we'll come up with a way of letting you see what features we are prioritizing.  Elliptic cone is an important feature.

louhz, 2025-03-24:
Thank you so much!
I will close this issue.

closed 2025-03-24 by louhz