google-deepmind/mujoco_warp #29: qfrc_actuator can be incorrect when computed by fwd_actuation (dense)

thowell, 2025-03-26:
`qfrc_actuator` can be incorrect when computed by `fwd_actuation` (dense).

reposting from #14:

simplifying actuation.xml by removing the third body + third joint + associated actuator from each branch will result in a test failure for `qfrc_actuator` in fwd_actuation_test.

```
<mujoco>
  <worldbody>
    <body pos="0 0 0">
      <joint name="hinge0" type="hinge" axis="1 0 0"/>
      <geom type="sphere" size=".1"/>
      <body>
        <joint name="hinge1" type="hinge" axis="1 0 0"/>
        <geom type="sphere" size=".1"/>
        <!-- <body>
          <joint name="hinge2" type="hinge" axis="1 0 0"/>
          <geom type="sphere" size=".1"/>
        </body> -->
      </body>
    </body>
    <body pos="1 0 0">
      <joint name="hinge3" type="hinge" axis="1 0 0"/>
      <geom type="sphere" size=".1"/>
      <body>
        <joint name="hinge4" type="hinge" axis="1 0 0"/>
        <geom type="sphere" size=".1"/>
        <!-- <body>
          <joint name="hinge5" type="hinge" axis="1 0 0"/>
          <geom type="sphere" size=".1"/>
        </body> -->
      </body>
    </body>
  </worldbody>
  <actuator>
    <motor joint="hinge0"/>
    <motor joint="hinge1"/>
    <!-- <motor joint="hinge2"/> -->
    <motor joint="hinge3"/>
    <motor joint="hinge4"/>
    <!-- <motor joint="hinge5"/> -->
  </actuator>
</mujoco>
```

adenzler-nvidia, 2025-04-04:
Identified the issue, it's a bug in warp. https://github.com/NVIDIA/warp/issues/622 tracked here, MR is in review.

adenzler-nvidia, 2025-04-23:
fixed in nightly warp, closing this issue.

closed 2025-04-23 by adenzler-nvidia