google-deepmind/mujoco_warp #47: Failed to compile tile_cholesky

arpit15, 2025-03-28:
Original [issue](https://github.com/NVIDIA/warp/issues/602) was originally filed in warp repo. The issue shows up when trying to run mjwarp-testspeed on a custom scene which consists of only boxes. Please refer to the original issue for system details. The scene is uploaded with this issue. 
Command used to reproduce the issue
```
mjwarp-testspeed --function=step --mjcf=test1.xml --batch_size=1 --nstep=1
```
[test1.txt](https://github.com/user-attachments/files/19496801/test1.txt)

thowell, 2025-03-28:
please try:
- `--solver=cg`
- `--is_sparse=True`

additionally, please note that box-box collisions are still in development #41

arpit15, 2025-03-28:
Thanks for the response. I added box-box from the MR in my repo. I tried the extra flags. I still get the same errors.

shi-eric, 2025-03-31:
This issue is two-fold:

- As we were discussing over in the Warp issue, tiles cannot be arbitrarily sized since they need to fit into shared memory for things like tile_cholesky to work. The `nv` (degrees of freedom?) parameter is 570 and this is resulting in a `570 x 570` tile (~1270 KiB for a tile of floats vs. 100 KiB of shared memory on my 3090 RTX). **On the Warp side, we need to make the issue clearer to the users rather than simply saying: `Failed to compile tile_cholesky.`**
- `mujoco_warp` will probably need to implement something to avoid creating such large tile sizes

adenzler-nvidia, 2025-04-04:
https://github.com/google-deepmind/mujoco_warp/pull/3 added a check to avoid large nvs. Please shout if you disagree with me closing this.

closed 2025-04-04 by adenzler-nvidia