google-deepmind/mujoco_warp #156: Benchmark fails due to recent commit using bleeding edge warp feature

Nate711, 2025-04-16:
I believe https://github.com/google-deepmind/mujoco_warp/commit/098721751ee67695e2c93728279771b86275e036 broke the benchmark because it uses `Device.sm_count` which was introduced very recently https://github.com/NVIDIA/warp/blob/main/CHANGELOG.md 

I'm using 
```
pip show warp-lang
Name: warp-lang
Version: 1.7.0
```

The error I get 
```
File "/home/nathankau/mujoco_warp/mujoco_warp/_src/solver.py", line 282, in _update_gradient
    sm_count = wp.get_device().sm_count
               ^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Device' object has no attribute 'sm_count'
```

btaba, 2025-04-16:
Hi @Nate711 , acknowledged about the version mismatch, you can try `pip install warp-lang --pre --upgrade -f https://pypi.nvidia.com/warp-lang/` for the time being

Nate711, 2025-04-16:
Thanks that worked

Nate711, 2025-04-16:
@btaba A bit unrelated, but do you guys have results numbers for the benchmark? I got this on my 5090:
```
(mjx2) nathankau@nathankau-3080:~/mujoco_warp$ mjwarp-testspeed --function=step --mjcf=test_data/humanoid/humanoid.xml --batch_size=8192
Warp 1.8.0.dev20250416 initialized:
   Git commit: 000a0d328b4d45b17c19b04834d0e765525d6ee5
   CUDA Toolkit 12.8, Driver 12.8
   Devices:
     "cpu"      : "x86_64"
     "cuda:0"   : "NVIDIA GeForce RTX 5090" (31 GiB, sm_120, mempool enabled)
   Kernel cache:
     /home/nathankau/.cache/warp/1.8.0.dev20250416
Model nbody: 17 nv: 27 ngeom: 20 is_sparse: None solver: Newton iterations: 10 ls_iterations: 20 linesearch: iterative
Data nworld: 8192 nconmax: 65536 njmax: 262144
Rolling out 1000 steps at dt = 0.005...
Module mujoco_warp._src.smooth.kinematics 6a2f45d load on device 'cuda:0' took 304.28 ms  (compiled)
Module mujoco_warp._src.smooth.com_pos 15b370a load on device 'cuda:0' took 244.02 ms  (compiled)
Module mujoco_warp._src.smooth.camlight 38456dc load on device 'cuda:0' took 246.27 ms  (compiled)
Module mujoco_warp._src.warp_util 86ece54 load on device 'cuda:0' took 426.13 ms  (compiled)
Module mujoco_warp._src.smooth.crb ddee9e8 load on device 'cuda:0' took 177.74 ms  (compiled)
Module mujoco_warp._src.smooth._factor_i_dense.tile_cholesky d8e08b6 load on device 'cuda:0' took 811.09 ms  (compiled)
Module mujoco_warp._src.collision_driver e8edae8 load on device 'cuda:0' took 314.17 ms  (compiled)
Module mujoco_warp._src.collision_convex a88b62f load on device 'cuda:0' took 45891.79 ms  (compiled)
Module mujoco_warp._src.collision_primitive 2f2cdea load on device 'cuda:0' took 5776.04 ms  (compiled)
Module mujoco_warp._src.collision_box ebdbcba load on device 'cuda:0' took 1695.33 ms  (compiled)
Module mujoco_warp._src.constraint 7b8064d load on device 'cuda:0' took 638.38 ms  (compiled)
Module mujoco_warp._src.smooth.transmission a7d9629 load on device 'cuda:0' took 189.72 ms  (compiled)
Module mujoco_warp._src.forward.fwd_velocity.actuator_velocity a1be27d load on device 'cuda:0' took 7531.95 ms  (compiled)
Module mujoco_warp._src.smooth.com_vel 4d538c7 load on device 'cuda:0' took 245.89 ms  (compiled)
Module mujoco_warp._src.passive.passive 546c695 load on device 'cuda:0' took 190.12 ms  (compiled)
Module mujoco_warp._src.smooth._rne_cacc_world b73b37b load on device 'cuda:0' took 111.99 ms  (compiled)
Module mujoco_warp._src.smooth._rne_cacc_forward 74e7591 load on device 'cuda:0' took 188.92 ms  (compiled)
Module mujoco_warp._src.smooth._rne_cfrc 7b2ac91 load on device 'cuda:0' took 123.70 ms  (compiled)
Module mujoco_warp._src.smooth._rne_cfrc_backward e2941c8 load on device 'cuda:0' took 125.29 ms  (compiled)
Module mujoco_warp._src.smooth.rne badb491 load on device 'cuda:0' took 124.51 ms  (compiled)
Module mujoco_warp._src.forward.fwd_actuation c08e9be load on device 'cuda:0' took 166.34 ms  (compiled)
Module mujoco_warp._src.forward.fwd_actuation.qfrc_actuator a2d1818 load on device 'cuda:0' took 3152.22 ms  (compiled)
Module mujoco_warp._src.forward.fwd_acceleration 096a6f6 load on device 'cuda:0' took 106.55 ms  (compiled)
Module mujoco_warp._src.support beadc28 load on device 'cuda:0' took 208.95 ms  (compiled)
Module mujoco_warp._src.smooth._solve_LD_dense.tile_cho_solve 0f44a33 load on device 'cuda:0' took 662.54 ms  (compiled)
Module mujoco_warp._src.solver._create_context dc3fd1a load on device 'cuda:0' took 158.81 ms  (compiled)
Module mujoco_warp._src.support.mul_m.tile_mul f9a7a0a load on device 'cuda:0' took 7171.66 ms  (compiled)
Module mujoco_warp._src.solver._update_constraint c5cbe3d load on device 'cuda:0' took 219.22 ms  (compiled)
Module mujoco_warp._src.solver._update_gradient 31d0a33 load on device 'cuda:0' took 1153.03 ms  (compiled)
Module mujoco_warp._src.solver._update_gradient 8b4370e load on device 'cuda:0' took 798.88 ms  (compiled)
Module mujoco_warp._src.solver._linesearch 88ef114 load on device 'cuda:0' took 264.88 ms  (compiled)
Module mujoco_warp._src.solver._linesearch_iterative 9b9f14e load on device 'cuda:0' took 349.29 ms  (compiled)
Module mujoco_warp._src.solver.solve 473acf5 load on device 'cuda:0' took 159.50 ms  (compiled)
Module mujoco_warp._src.forward._advance 64e1261 load on device 'cuda:0' took 226.18 ms  (compiled)

Summary for 8192 parallel rollouts

 Total JIT time: 80.81 s
 Total simulation time: 7.31 s
 Total steps per second: 1,120,357
 Total realtime factor: 5,601.79 x
 Total time per step: 892.57 ns
```

btaba, 2025-04-16:
Hi @Nate711 deferring to @erikfrey for current metrics!

erikfrey, 2025-04-16:
There's a bit too much in the air right now to report up to date benchmarks.  We shared some benchmarks in our [GTC announce](https://www.nvidia.com/en-us/on-demand/session/gtc25-S72709/), feel free to check those out.

In particular, the `humanoid.xml` that you benchmarked was recently adjusted in anticipation of #78 landing, so we purposefully made `humanoid.xml` slower than it needs to be so we can make it faster again      :-)

closed 2025-04-27 by Nate711