google-deepmind/mujoco_warp #163: Box<->Capsule Collision

Andrew-Luo1, 2025-04-18:
**Setup**

Simulating scene_terrain.xml from https://github.com/google-deepmind/mujoco_warp/pull/162 using mujoco warp's viewer. 

**The Issue**

Even with nconmax set to 512*1024 in put_data, n_worlds = 1 and shrinking down scene_terrain.xml down to a 3x3 grid of subterrains (versus the 10x20 default), collisions with pyramids don't look correct. (This doesn't happen in cmujoco).

[image]

[image]

camevor, 2025-04-22:
Hi @Andrew-Luo1, the capsule-box implementation is still under review (#135). Could you try your scene with that branch?
I just tried it and capsule-box looks good to me, but I ran into some box-box collision errors that I'm looking into right now.

adenzler-nvidia, 2025-04-23:
@Andrew-Luo1 while you're at it - Chris would love to have some feedback on that PR.

Andrew-Luo1, 2025-04-23:
Hi @camevor and @adenzler-nvidia something seems to have broken my dev environment between when I PR'd this terrain and the current upstream on the capsule-box branch, i.e. I can't run the viewer. Actually I've recently begun a stint in the LLM world and will have very limited time to work on this project. Apologies and best of luck!

adenzler-nvidia, 2025-05-05:
@camevor - can we close this once #208 is merged?

camevor, 2025-05-05:
Yes, running the apollo scene again with the new box-box and capsule-box, the behavior looks good.

closed 2025-05-13 by camevor