google-deepmind/mujoco_warp #177: Segmentation fault

SiweiCui, 2025-04-24:
Hi! I've got a Segmentation fault. 

When I run `pytest`, everything works just fine.

```
================================================================== test session starts ==================================================================
platform linux -- Python 3.10.16, pytest-8.3.5, pluggy-1.5.0
rootdir: /cpfs04/user/cuisiwei/mujoco_warp
configfile: pyproject.toml
collected 141 items                                                                                                                                     

contrib/kernel_analyzer/kernel_analyzer/ast_analyzer_test.py ............                                                                         [  8%]
mujoco_warp/_src/broadphase_test.py ..                                                                                                            [  9%]
mujoco_warp/_src/collision_driver_test.py ......................                                                                                  [ 25%]
mujoco_warp/_src/constraint_test.py .......................                                                                                       [ 41%]
mujoco_warp/_src/forward_test.py ............                                                                                                     [ 50%]
mujoco_warp/_src/io_test.py ...........                                                                                                           [ 58%]
mujoco_warp/_src/math_test.py .......                                                                                                             [ 63%]
mujoco_warp/_src/passive_test.py .                                                                                                                [ 63%]
mujoco_warp/_src/sensor_test.py ..                                                                                                                [ 65%]
mujoco_warp/_src/smooth_test.py ....................                                                                                              [ 79%]
mujoco_warp/_src/solver_test.py ..........                                                                                                        [ 86%]
mujoco_warp/_src/support_test.py ...................                                                                                              [100%]

============================================================ 141 passed in 382.79s (0:06:22) ============================================================
```

But when I run `mjwarp-testspeed --function=step --mjcf=test_data/humanoid/humanoid.xml --batch_size=8192 --event_trace=True`, I get:

```
Warp 1.8.0.dev20250423 initialized:
   Git commit: 874bf60a2586eb4f4e55a04f57cad7b4f2431981
   CUDA Toolkit 12.8, Driver 12.4
   Devices:
     "cpu"      : "x86_64"
     "cuda:0"   : "NVIDIA A800-SXM4-80GB" (79 GiB, sm_80, mempool enabled)
   Kernel cache:
     /root/.cache/warp/1.8.0.dev20250423
Model nbody: 17 nv: 27 ngeom: 20 is_sparse: None solver: Newton iterations: 10 ls_iterations: 20 linesearch: iterative
Data nworld: 8192 nconmax: 65536 njmax: 262144
Rolling out 1000 steps at dt = 0.005...
Module mujoco_warp._src.smooth.kinematics 29b6e41 load on device 'cuda:0' took 0.62 ms  (cached)
Module mujoco_warp._src.smooth.com_pos b8f75b8 load on device 'cuda:0' took 0.43 ms  (cached)
Module mujoco_warp._src.smooth.camlight fe3edd3 load on device 'cuda:0' took 0.39 ms  (cached)
Module mujoco_warp._src.warp_util 86ece54 load on device 'cuda:0' took 0.31 ms  (cached)
Module mujoco_warp._src.smooth.crb 629bb36 load on device 'cuda:0' took 0.29 ms  (cached)
Module mujoco_warp._src.smooth._factor_i_dense.tile_cholesky dc1218b load on device 'cuda:0' took 10.65 ms  (cached)
Module mujoco_warp._src.collision_driver bb38562 load on device 'cuda:0' took 0.33 ms  (cached)
Module mujoco_warp._src.collision_convex f096622 load on device 'cuda:0' took 1.57 ms  (cached)
Module mujoco_warp._src.collision_primitive d064999 load on device 'cuda:0' took 0.74 ms  (cached)
Module mujoco_warp._src.collision_box 0bc838f load on device 'cuda:0' took 0.56 ms  (cached)
Module mujoco_warp._src.constraint c09c9ba load on device 'cuda:0' took 0.56 ms  (cached)
Module mujoco_warp._src.constraint 01df401 load on device 'cuda:0' took 0.54 ms  (cached)
Module mujoco_warp._src.smooth.transmission f0b23a9 load on device 'cuda:0' took 13.44 ms  (cached)
Module mujoco_warp._src.forward.fwd_velocity.actuator_velocity 3c25edc load on device 'cuda:0' took 34.20 ms  (cached)
Module mujoco_warp._src.smooth.com_vel 260eea5 load on device 'cuda:0' took 0.38 ms  (cached)
Module mujoco_warp._src.passive.passive 2f3cc09 load on device 'cuda:0' took 0.36 ms  (cached)
Module mujoco_warp._src.smooth._rne_cacc_world 0c33f66 load on device 'cuda:0' took 0.27 ms  (cached)
Module mujoco_warp._src.smooth._rne_cacc_forward ac84993 load on device 'cuda:0' took 0.25 ms  (cached)
Module mujoco_warp._src.smooth._rne_cfrc 803430b load on device 'cuda:0' took 0.26 ms  (cached)
Module mujoco_warp._src.smooth._rne_cfrc_backward 45381fc load on device 'cuda:0' took 0.26 ms  (cached)
Module mujoco_warp._src.smooth.rne 95a0a4c load on device 'cuda:0' took 0.24 ms  (cached)
Module mujoco_warp._src.forward.fwd_actuation dd68fd9 load on device 'cuda:0' took 0.31 ms  (cached)
Module mujoco_warp._src.forward.fwd_actuation.qfrc_actuator 5007f2e load on device 'cuda:0' took 34.67 ms  (cached)
Module mujoco_warp._src.forward.fwd_acceleration a2d7128 load on device 'cuda:0' took 0.33 ms  (cached)
Module mujoco_warp._src.support 628a112 load on device 'cuda:0' took 0.33 ms  (cached)
Module mujoco_warp._src.smooth._solve_LD_dense.tile_cho_solve beebfb3 load on device 'cuda:0' took 13.37 ms  (cached)
Module mujoco_warp._src.solver._create_context baf70d3 load on device 'cuda:0' took 0.35 ms  (cached)
Module mujoco_warp._src.support.mul_m.tile_mul b62b82a load on device 'cuda:0' took 34.47 ms  (cached)
Module mujoco_warp._src.solver._update_constraint cf8279a load on device 'cuda:0' took 0.43 ms  (cached)
[... 48 lines cut ...]
packaging           25.0
pip                 25.0.1
pluggy              1.5.0
PyOpenGL            3.1.9
pytest              8.3.5
setuptools          75.8.0
tomli               2.2.1
typing_extensions   4.13.2
warp-lang           1.8.0.dev20250423
wheel               0.45.1
zipp                3.21.0
```

My python version is 3.10, and my OS is Ubuntu 22.04

```
Python 3.10.16 (main, Dec 11 2024, 16:24:50) [GCC 11.2.0]
```

```
PRETTY_NAME="Ubuntu 22.04.5 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.5 LTS (Jammy Jellyfish)"
```

adenzler-nvidia, 2025-04-24:
Unfortunately sometimes the error reporting is not great and things just show up as a segfault like this. A couple of questions:
* are you using the latest nightly warp version?
* does it make a difference if you upgrade your CUDA driver to a version that is lining up with your 12.8 toolkit? I think that would be a 570+ driver version
* can you try adding wp.config.mode = "debug" and see whether you get a different error?
* can you try adding wp.config.verify_cuda = True and see whether you get a different error?

SiweiCui, 2025-05-22:
Upgrading CUDA driver works. Thank you!