google-deepmind/mujoco_warp #250: Apollo heightfield benchmark scene hangs when loading

erikfrey, 2025-05-14:
The hang comes from [io.py](https://github.com/google-deepmind/mujoco_warp/blob/main/mujoco_warp/_src/io.py#L223):

```py
  for geom1, geom2 in zip(*np.triu_indices(mjm.ngeom, k=1)):  # k=1 skip diagonal
    ...
```

In this scene, `ngeom` = 5,290.  Iterating over 5,290^2 nested for loop is taking ~5 minutes.

yuvaltassa, 2025-05-14:
Is this not where we should try to flex Warp's allegedly good BVH implementation?

thowell, 2025-05-14:
setup:
- mujoco warp main
- warp 1.8.0.dev20250513
- NVIDIA RTX A4000
- CUDA Toolkit 12.8, Driver 12.8

`mjwarp-viewer --mjcf=mujoco_warp/test_data/apptronik_apollo/scene_terrain.xml`

first run:

```
Compiling the model physics step...
Module mujoco_warp._src.smooth 7c923ff load on device 'cuda:0' took 2115.64 ms  (compiled)
Module mujoco_warp._src.smooth._tile_cholesky 6aea4a9 load on device 'cuda:0' took 893.90 ms  (compiled)
Module mujoco_warp._src.collision_driver 1032a63 load on device 'cuda:0' took 301.57 ms  (compiled)
Module mujoco_warp._src.collision_convex 3431190 load on device 'cuda:0' took 41718.05 ms  (compiled)
Module mujoco_warp._src.collision_primitive bf01bc1 load on device 'cuda:0' took 3524.40 ms  (compiled)
Module mujoco_warp._src.constraint 60e1830 load on device 'cuda:0' took 1367.77 ms  (compiled)
Module mujoco_warp._src.forward._tile_actuator_velocity fc2c962 load on device 'cuda:0' took 12845.66 ms  (compiled)
Module mujoco_warp._src.passive 72ef6af load on device 'cuda:0' took 555.35 ms  (compiled)
Module mujoco_warp._src.forward 59245ec load on device 'cuda:0' took 629.20 ms  (compiled)
Module mujoco_warp._src.forward._tile_qfrc_actuator cecb4ad load on device 'cuda:0' took 4650.19 ms  (compiled)
Module mujoco_warp._src.support b3274f5 load on device 'cuda:0' took 371.10 ms  (compiled)
Module mujoco_warp._src.smooth._tile_cho_solve bf3714b load on device 'cuda:0' took 698.46 ms  (compiled)
Module mujoco_warp._src.solver f65c0e9 load on device 'cuda:0' took 1828.08 ms  (compiled)
Module mujoco_warp._src.support.mul_m_dense 0887d89 load on device 'cuda:0' took 12145.52 ms  (compiled)
Module mujoco_warp._src.solver.update_gradient_cholesky 33da2cb load on device 'cuda:0' took 289.98 ms  (compiled)
Compilation took 84.66462135314941s.
```

second run

```
Compiling the model physics step...
Module mujoco_warp._src.smooth 7c923ff load on device 'cuda:0' took 5.60 ms  (cached)
Module mujoco_warp._src.smooth._tile_cholesky 6aea4a9 load on device 'cuda:0' took 9.88 ms  (cached)
Module mujoco_warp._src.collision_driver 1032a63 load on device 'cuda:0' took 0.91 ms  (cached)
Module mujoco_warp._src.collision_convex 3431190 load on device 'cuda:0' took 9.52 ms  (cached)
Module mujoco_warp._src.collision_primitive bf01bc1 load on device 'cuda:0' took 1.43 ms  (cached)
Module mujoco_warp._src.constraint 60e1830 load on device 'cuda:0' took 1.51 ms  (cached)
Module mujoco_warp._src.forward._tile_actuator_velocity fc2c962 load on device 'cuda:0' took 29.52 ms  (cached)
Module mujoco_warp._src.passive 72ef6af load on device 'cuda:0' took 0.86 ms  (cached)
Module mujoco_warp._src.forward 59245ec load on device 'cuda:0' took 1.26 ms  (cached)
Module mujoco_warp._src.forward._tile_qfrc_actuator cecb4ad load on device 'cuda:0' took 30.77 ms  (cached)
Module mujoco_warp._src.support b3274f5 load on device 'cuda:0' took 0.83 ms  (cached)
Module mujoco_warp._src.smooth._tile_cho_solve bf3714b load on device 'cuda:0' took 11.69 ms  (cached)
Module mujoco_warp._src.solver f65c0e9 load on device 'cuda:0' took 2.41 ms  (cached)
Module mujoco_warp._src.support.mul_m_dense 0887d89 load on device 'cuda:0' took 28.65 ms  (cached)
Module mujoco_warp._src.solver.update_gradient_cholesky 33da2cb load on device 'cuda:0' took 17.00 ms  (cached)
Compilation took 0.8503944873809814s.
```

thowell, 2025-05-26:
Closing this issue since #293 has been merged.

closed 2025-05-26 by thowell