google-deepmind/mujoco_warp #258: Convex mesh collision doesn't work

vreutskyy, 2025-05-15:
Here's a scene with a ball falling onto a convex mesh box. The ball flies away instead of falling down.
```
<mujoco>
  <asset>
    <!-- Simple box mesh, size will be scaled by the geom size -->
    <mesh name="box_mesh" vertex="1 1 0  1 1 -1  1 -1 0  1 -1 -1  -1 1 0  -1 1 -1  -1 -1 0  -1 -1 -1"/>
    <!-- -->
  </asset>
  <worldbody>
    <!-- Ground plane -->
    <geom type="plane" size="2 2 0.1" pos="0 0 -0.3" rgba="0.8 0.8 0.8 1"/>
    <geom type="mesh" mesh="box_mesh" pos="0 0 0" rgba="0.4 0.6 0.4 1"/>
    <body pos="0 0.1 0.3">
      <freejoint/>
      <geom type="sphere" size="0.1 0.1 0.1" rgba="0.8 0.2 0.2 1"/>
    </body>
  </worldbody>
</mujoco>
```

kbayes, 2025-05-16:
I ported over the GJK/EPA from MuJoCo C over. This should automatically get resolved with that implementation.

vreutskyy, 2025-05-16:
Is it going to happen soon?

kbayes, 2025-05-16:
Yes, a stripped down version has been ported over, just debugging right now. I'll have the GJK portion PR out on Monday. This will be enough to do sphere-mesh collisions.

vreutskyy, 2025-05-20:
The bug is actually in passing `geom_size[worldid]` to `_geom()` instead of just `geom_size`

here:
https://github.com/google-deepmind/mujoco_warp/blob/63cf702246433bb565de915290769cb42afc9fb8/mujoco_warp/_src/collision_convex.py#L787

and here:
https://github.com/google-deepmind/mujoco_warp/blob/63cf702246433bb565de915290769cb42afc9fb8/mujoco_warp/_src/collision_convex.py#L800

I fixed it in my branch

closed 2025-05-21 by adenzler-nvidia