google-deepmind/mujoco_warp #260: Performance issues when comparing against Mujoco C?

gpu-daal, 2025-05-16:
I've run aloha_pot example under Mujoco C and Mujoco Warp, and while I do see almost 100% GPU usage when using Warp version, the performance tanks quite a lot (tested on RTX 4070 SUPER with 12GB VRAM)

MJC-Warp with ls_parallel flag on: almost 50 FPS with V-sync off on, the arms are very unstable and move on their own, the pot sinks through the table. At some point everything except for the table disappears (including the light for some reason).
MJC: almost 400 FPS with V-sync off, everything is smooth and stable, the arms respond pretty quickly.

I assume the physics tick on the GPU is just not long enough to process everything. I've run an example of my own where a little cube just falls on a table, and with MJC-Warp it would (very slowly) fall and phase through the table, while MJC correctly resolves collisions. 

I also thought that maybe Warp is more suited to run multiple environments/robots, but it kind of defeats the purpose of aloha_pot example. I might also be a little too early to compare the performance since this is Alpha version.

Is there perhaps anything I am doing wrong? Thank you!

erikfrey, 2025-05-16:
Hi @gpu-daal - it's too soon to interpret any meaningful perf stats from the benchmarks.  The benchmarks are an indication of the kinds of scenes we expect to simulate well, but for now there's still bugs and missing features.

Give us a few weeks to graduate from Alpha to Beta, get things working correctly, and then we'll start measuring and reporting benchmarks.

kbayes, 2025-06-30:
Hi @gpu-daal, we're currently at beta if you want to give things another try.

gpu-daal, 2025-06-30:
@kbayes I've just tested aloha pot again on a different GPU. Though the simulation is more stable and FPS is higher than a month ago, it's still worse than CPU version. I guess it is due to rendering and running only 1 environment? Running viewer with multiple nworld results in NotImplementedError.

I also tested kitchen environment with Panda robot, and the simulation is way too slow. Robot responds very slowly. If you put a cube (simple free falling cube with joint of type free) over a table, cube will fall (very slowly) through the table. I guess SAP collision detection cannot handle more than 100 objects?

Anyways, congrats on making it to beta! Excited to see benchmark results and RL integration.