google-deepmind/mujoco_warp #425: Crash during training: CUDA_ERROR_STREAM_CAPTURE_INVALIDATED  vs. convex_narrowphase

johnm, 2025-06-28:
Tested against the latest version of mujoco_warp as of this morning, 13bcf2e.

Included is a standalone python program using mujoco_warp. With the unpatched installation as per the README instructions, running it crashes *after* training Step 0. Complete logs included below.

If I comment out the call to `convex_narrowphase()` in `collision_driver.py`:
```
 # TODO(team): we should reject far-away contacts in the narrowphase instead of constraint
 #             partitioning because we can move some pressure of the atomics
 # TODO(team) switch between collision functions and GJK/EPA here
 # convex_narrowphase(m, d)
 primitive_narrowphase(m, d)
 sdf_narrowphase(m, d)
```
then the training will complete successfully. Logs for that following below.

Would be happy to learn it's something silly on my end but I think this is a deeper issue.

Logs before patch--crash:
```
% python3 train_apollo_locomotion.py --num_timesteps 10000000
Warp 1.8.0.dev20250627 initialized:
  Git commit: 93ac131e00300bf192738c1589a93281c4cfdaa1
  CUDA Toolkit 12.8, Driver 12.4
  Devices:
    "cpu"      : "x86_64"
    "cuda:0"   : "NVIDIA GeForce RTX 4090" (24 GiB, sm_89, mempool enabled)
  Kernel cache:
    /home/johnm/.cache/warp/1.8.0.dev20250627
INFO:__main__:============================================================
INFO:__main__:Starting Apptronik Apollo Locomotion Training
INFO:__main__:============================================================
INFO:__main__:Training timesteps: 10,000,000
INFO:__main__:Output directory: ./apollo_model_output
INFO:__main__:Learning rate: 0.0003
INFO:__main__:Number of evaluations: 10
INFO:__main__:Batch size (worlds): 8192
INFO:__main__:JAX devices: [CudaDevice(id=0)]
INFO:__main__:JAX device count: 1
INFO:__main__:Setting up MuJoCo model and Warp data structures...
INFO:__main__:Pre-warming Warp kernels...
Module mujoco_warp._src.smooth 90d539d load on device 'cuda:0' took 4.91 ms  (cached)
Module mujoco_warp._src.collision_driver 7252014 load on device 'cuda:0' took 0.13 ms  (cached)
Module mujoco_warp._src.collision_convex 85ab1c2 load on device 'cuda:0' took 0.24 ms  (cached)
Module mujoco_warp._src.collision_convex ece7377 load on device 'cuda:0' took 0.19 ms  (cached)
Module mujoco_warp._src.collision_convex 25bc740 load on device 'cuda:0' took 0.19 ms  (cached)
Module mujoco_warp._src.collision_primitive 9f8e935 load on device 'cuda:0' took 0.15 ms  (cached)
Module mujoco_warp._src.collision_sdf 82cee96 load on device 'cuda:0' took 0.12 ms  (cached)
Module mujoco_warp._src.constraint 2a7cd9c load on device 'cuda:0' took 0.19 ms  (cached)
Module mujoco_warp._src.forward._actuator_velocity_sparse a1d98d3 load on device 'cuda:0' took 0.12 ms  (cached)
Module mujoco_warp._src.passive c5447c7 load on device 'cuda:0' took 0.13 ms  (cached)
Module mujoco_warp._src.forward fb16fa3 load on device 'cuda:0' took 0.17 ms  (cached)
Module mujoco_warp._src.support b8eb504 load on device 'cuda:0' took 0.11 ms  (cached)
Module mujoco_warp._src.solver 0f5d3f0 load on device 'cuda:0' took 0.31 ms  (cached)
Module mujoco_warp._src.solver.update_gradient_cholesky 33da2cb load on device 'cuda:0' took 12.49 ms  (cached)
Module mujoco_warp._src.solver.linesearch_jv_fused d6ee3c1 load on device 'cuda:0' took 0.12 ms  (cached)
Module mujoco_warp._src.collision_convex 85ab1c2 load on device 'cuda:0' took 0.22 ms  (cached)
Module mujoco_warp._src.collision_convex ece7377 load on device 'cuda:0' took 0.20 ms  (cached)
Module mujoco_warp._src.collision_convex 25bc740 load on device 'cuda:0' took 0.20 ms  (cached)
Module mujoco_warp._src.collision_convex 85ab1c2 load on device 'cuda:0' took 0.22 ms  (cached)
Module mujoco_warp._src.collision_convex ece7377 load on device 'cuda:0' took 0.18 ms  (cached)
Module mujoco_warp._src.collision_convex 25bc740 load on device 'cuda:0' took 0.19 ms  (cached)
Module mujoco_warp._src.collision_convex 85ab1c2 load on device 'cuda:0' took 0.24 ms  (cached)
Module mujoco_warp._src.collision_convex ece7377 load on device 'cuda:0' took 0.18 ms  (cached)
Module mujoco_warp._src.collision_convex 25bc740 load on device 'cuda:0' took 0.18 ms  (cached)
Module mujoco_warp._src.collision_convex 85ab1c2 load on device 'cuda:0' took 0.21 ms  (cached)
Module mujoco_warp._src.collision_convex ece7377 load on device 'cuda:0' took 0.17 ms  (cached)
Module mujoco_warp._src.collision_convex 25bc740 load on device 'cuda:0' took 0.19 ms  (cached)
Module mujoco_warp._src.collision_convex 85ab1c2 load on device 'cuda:0' took 0.20 ms  (cached)
Module mujoco_warp._src.collision_convex ece7377 load on device 'cuda:0' took 0.18 ms  (cached)
Module mujoco_warp._src.collision_convex 25bc740 load on device 'cuda:0' took 0.18 ms  (cached)
[... 1005 lines cut ...]
       # Performance summary
       if training_data['throughput']:
           avg_throughput = np.mean(training_data['throughput'][1:])  # Skip first measurement
           max_throughput = np.max(training_data['throughput'][1:])
           [logger.info](http://logger.info/)(f"Average throughput: {avg_throughput:.1f} steps/sec")
           [logger.info](http://logger.info/)(f"Peak throughput: {max_throughput:.1f} steps/sec")

       # Save the model
       save_model(params, {
           'training_metrics': metrics,
           'training_data': training_data,
           'config': dict(ppo_config),
           'args': vars(args)
       }, args.output_dir)

       [logger.info](http://logger.info/)("Training script completed successfully!")

   except Exception as e:
       logger.error(f"Training failed with error: {str(e)}")
       logger.exception("Full traceback:")
       sys.exit(1)

if __name__ == "__main__":
   main()
```

erikfrey, 2025-06-30:
Thanks @johnm  - I suspect this is fixed by #426 but I'll see if I can repro and confirm.

johnm, 2025-07-02:
> Thanks [@johnm](https://github.com/johnm) - I suspect this is fixed by [#426](https://github.com/google-deepmind/mujoco_warp/pull/426) but I'll see if I can repro and confirm.

Just pulled the latest and tested without the local hack and training did indeed run without crashing! Yea!

johnm, 2025-07-02:
But a new time suck was introduced...

The first time I ran after updating mujoco-warp, this calls took a **long** time:
```
Module mujoco_warp._src.collision_convex.ccd_kernel_builder 1f96776 load on device 'cuda:0' took 44489.26 ms  (compiled)
Module mujoco_warp._src.collision_convex.ccd_kernel_builder cf58767 load on device 'cuda:0' took 64367.39 ms  (compiled)
```

The second time was:
```
Module mujoco_warp._src.collision_convex.ccd_kernel_builder 1f96776 load on device 'cuda:0' took 0.35 ms  (cached)
Module mujoco_warp._src.collision_convex.ccd_kernel_builder cf58767 load on device 'cuda:0' took 0.44 ms  (cached)
```

How should I think about that compilation time? I.e., when should I expect those long compilation times? Once per ???

Thanks!

Here's the full log of the first run:
```
python3 train_apollo_locomotion.py --num_timesteps 10000000
Warp 1.9.0.dev20250702 initialized:
   Git commit: 2831b62ef0b2ce9641885e6de5f082853517790b
   CUDA Toolkit 12.8, Driver 12.4
   Devices:
     "cpu"      : "x86_64"
     "cuda:0"   : "NVIDIA GeForce RTX 4090" (24 GiB, sm_89, mempool enabled)
   Kernel cache:
     /home/johnm/.cache/warp/1.9.0.dev20250702
INFO:__main__:============================================================
INFO:__main__:Starting Apptronik Apollo Locomotion Training
INFO:__main__:============================================================
INFO:__main__:Training timesteps: 10,000,000
INFO:__main__:Output directory: ./apollo_model_output
INFO:__main__:Learning rate: 0.0003
INFO:__main__:Number of evaluations: 10
INFO:__main__:Batch size (worlds): 8192
INFO:__main__:JAX devices: [CudaDevice(id=0)]
INFO:__main__:JAX device count: 1
INFO:__main__:Setting up MuJoCo model and Warp data structures...
INFO:__main__:Pre-warming Warp kernels...
Module mujoco_warp._src.smooth 90d539d load on device 'cuda:0' took 4.91 ms  (cached)
Module mujoco_warp._src.collision_driver 7252014 load on device 'cuda:0' took 0.17 ms  (cached)
Module mujoco_warp._src.collision_convex.ccd_kernel_builder 8732a6c load on device 'cuda:0' took 0.27 ms  (cached)
Module mujoco_warp._src.collision_convex.ccd_kernel_builder 1f96776 load on device 'cuda:0' took 44489.26 ms  (compiled)
Module mujoco_warp._src.collision_convex.ccd_kernel_builder cf58767 load on device 'cuda:0' took 64367.39 ms  (compiled)
Module mujoco_warp._src.collision_primitive 9f8e935 load on device 'cuda:0' took 0.19 ms  (cached)
Module mujoco_warp._src.constraint 2a7cd9c load on device 'cuda:0' took 0.21 ms  (cached)
Module mujoco_warp._src.forward._actuator_velocity_sparse a1d98d3 load on device 'cuda:0' took 124.93 ms  (compiled)
Module mujoco_warp._src.passive c5447c7 load on device 'cuda:0' took 0.14 ms  (cached)
Module mujoco_warp._src.forward fb16fa3 load on device 'cuda:0' took 0.17 ms  (cached)
Module mujoco_warp._src.support b8eb504 load on device 'cuda:0' took 0.12 ms  (cached)
Module mujoco_warp._src.solver 0f5d3f0 load on device 'cuda:0' took 0.32 ms  (cached)
Module mujoco_warp._src.solver.update_gradient_cholesky 33da2cb load on device 'cuda:0' took 1693.39 ms  (compiled)
Module mujoco_warp._src.solver.linesearch_jv_fused d6ee3c1 load on device 'cuda:0' took 133.89 ms  (compiled)
INFO:__main__:Warp kernels pre-warmed
INFO:__main__:MuJoCo model and Warp data structures initialized successfully
INFO:__main__:MuJoCo Warp step function created successfully
INFO:__main__:Joystick environment created successfully
INFO:__main__:Action size: 19
INFO:__main__:Starting PPO training...
INFO:__main__:Initializing JAX compilation (this may take several minutes)...
Module __main__ e8ce936 load on device 'cuda:0' took 85.80 ms  (compiled)
INFO:__main__:Step 0: reward=-4.960 ± 1.623
INFO:__main__:  Performance: 0.0 steps/sec, Time: 23.6s
INFO:__main__:  eval/walltime: 21.227692
INFO:__main__:  eval/avg_episode_length: 44.716675
INFO:__main__:  eval/std_episode_length: 13.077105
INFO:__main__:  eval/epoch_eval_time: 21.227692
INFO:__main__:  eval/sps: 385911.013544
[... 49 lines cut ...]
INFO:__main__:Step 9175040: reward=-0.435 ± 0.212
INFO:__main__:  Performance: 48484.8 steps/sec, Time: 23.7s
INFO:__main__:  eval/walltime: 183.230399
INFO:__main__:  eval/avg_episode_length: 55.047485
INFO:__main__:  eval/std_episode_length: 12.447498
INFO:__main__:  eval/epoch_eval_time: 20.098200
INFO:__main__:  eval/sps: 407598.688751
INFO:__main__:Step 10321920: reward=-0.307 ± 0.205
INFO:__main__:  Performance: 48276.4 steps/sec, Time: 23.8s
INFO:__main__:  eval/walltime: 203.429823
INFO:__main__:  eval/avg_episode_length: 60.285645
INFO:__main__:  eval/std_episode_length: 13.382470
INFO:__main__:  eval/epoch_eval_time: 20.199424
INFO:__main__:  eval/sps: 405556.118258
INFO:__main__:============================================================
INFO:__main__:Training completed successfully!
INFO:__main__:Total training time: 239.74 seconds (0.07 hours)
INFO:__main__:Final episode reward: -0.307
INFO:__main__:============================================================
INFO:__main__:Average throughput: 47877.6 steps/sec
INFO:__main__:Peak throughput: 48484.8 steps/sec
INFO:__main__:Model parameters saved to ./apollo_model_output/model_params.pkl
INFO:__main__:Training metrics saved to ./apollo_model_output/training_metrics.pkl
INFO:__main__:Training script completed successfully!
```

adenzler-nvidia, 2025-07-04:
Unfortunately we have very long compilation times for the convex collision code at the moment. We are aware of this and looking into improving things.

The cache fortunately avoids a recompile most of the time but it will be invalidated if the kernels themselves change or you update your version of warp.