google-deepmind/mujoco_warp #441: Bug: Contact Force Jumps in MjWarp

jc-bao, 2025-07-01:
I've encountered an issue in MjWarp (currently in beta) where the contact forces, specifically the support force of an object, appear to jump erratically during a simple motion simulation. This behavior is not observed when running the same simulation using MuJoCo CPU or MJX, where the contact forces are smooth and as expected.

I am unsure if this is an expected behavior of MjWarp in its current beta stage, a misconfiguration on my part, or if there's a need to recompile the graph periodically.

MjWarp offers significant performance improvements for my use case, so any guidance or information on this would be greatly appreciated!

```python
def visualize(
    dt_sim: float = 0.005,
    dt_ctrl: float = 0.005,
    model_path: str = "../assets/inspire/scene_bimanual_wipe.xml",
    data_path: str = "../../outputs/inspire_bimanual_wipe.npz",
) -> None:
    """
    Main function to load data, set up the simulation, and run the viewer.

    Args:
        dt_sim: Simulation timestep.
        dt_ctrl: Control timestep.
        model_path: Path to the MuJoCo XML model file.
        data_path: Path to the NPZ file containing trajectory data (qpos, qvel, ctrl).
    """
    # --- 1. Load Trajectory Data ---
    print(f"Loading trajectory data from: {data_path}")
    if not os.path.exists(data_path):
        raise FileNotFoundError(f"Data file not found at {data_path}")
    raw_data = np.load(data_path)

    qpos_list = raw_data["qpos"]
    qvel_list = raw_data["qvel"]
    if "ctrl" in raw_data:
        ctrl_list = raw_data["ctrl"]
    else:
        # Fallback if 'ctrl' is not in the data file
        print("Warning: 'ctrl' data not found, using 'qpos' as a fallback for control.")
        ctrl_list = qpos_list[:, :-14]  # Assuming the model has 14 DoFs for object2

    # --- 2. Initialize MuJoCo (for viewer and initial state) ---
    print(f"Loading MuJoCo model from: {model_path}")
    if not os.path.exists(model_path):
        raise FileNotFoundError(f"Model file not found at {model_path}")
    mjm = mujoco.MjModel.from_xml_path(model_path)
    mjm.opt.timestep = dt_sim
    mjm.opt.iterations = 10
    mjm.opt.ls_iterations = 20
    mjm.opt.o_solimp = [0.9, 0.95, 0.001, 0.5, 2]
    mjd = mujoco.MjData(mjm)

    # Set initial state from the first frame of the trajectory
    mjd.qpos[:] = qpos_list[0]
    mjd.qvel[:] = qvel_list[0]
    mjd.ctrl[:] = ctrl_list[0]
    mujoco.mj_forward(mjm, mjd)

    # --- 3. Initialize MJWarp and Compile Step ---
    print("Putting model and data onto Warp device (GPU)...")
    m = mjwarp.put_model(mjm)
    d = mjwarp.put_data(mjm, mjd)

    print("JIT-compiling the model physics step with CUDA graphs...")
    start_time = time.time()
    graph = _compile_step(m, d)
    elapsed = time.time() - start_time
    print(f"Compilation took {elapsed:.3f}s.")

    # --- 4. Launch Viewer and Run Simulation Loop ---
    print("Launching passive viewer...")
    viewer = mujoco.viewer.launch_passive(mjm, mjd)
    with viewer:
        rate_limiter = RateLimiter(1.0 / dt_sim)

        while viewer.is_running():
            # Determine the control index based on simulation time, looping the trajectory
            ctrl_idx = int(mjd.time / dt_ctrl) % qpos_list.shape[0]
            mjd.ctrl[:] = ctrl_list[ctrl_idx]

            # --- MJWarp Simulation Step ---
            # a. Copy current state from MuJoCo (CPU) to MJWarp (GPU)
            # This is necessary to keep the Warp simulation state in sync with any
            # potential external changes (like control updates).
            wp.copy(d.ctrl, wp.array([mjd.ctrl.astype(np.float32)]))
            wp.copy(d.act, wp.array([mjd.act.astype(np.float32)]))
            wp.copy(d.xfrc_applied, wp.array([mjd.xfrc_applied.astype(np.float32)]))
            wp.copy(d.qpos, wp.array([mjd.qpos.astype(np.float32)]))
            wp.copy(d.qvel, wp.array([mjd.qvel.astype(np.float32)]))
            wp.copy(d.time, wp.array([mjd.time], dtype=wp.float32))

            # b. Execute the pre-compiled physics step on the GPU
            wp.capture_launch(graph)

            # c. Copy results from MJWarp (GPU) back to MuJoCo (CPU) for rendering
            mjwarp.get_data_into(mjd, mjm, d)

            # d. Sync the viewer to render the new state
            viewer.sync()

            # e. Maintain the simulation rate
            rate_limiter.sleep()

```

https://github.com/user-attachments/assets/583b8a1b-a295-48fd-b926-7d97231c7a9d

thowell, 2025-07-04:
@jc-bao thank you for sharing the repro!

the contact forces from experiments with the environment `scene_bimanual_wipe.xml` in the mujoco warp viewer appear to be reasonable

currently need to uncomment the linked line and the next line to visualize the contact forces
https://github.com/google-deepmind/mujoco_warp/blob/2c43d29166ee7aad5f63ddf08e840ce1af875e7b/mujoco_warp/_src/io.py#L1525

https://github.com/user-attachments/assets/b692c4f8-fafd-4fb2-b405-96ec045170c2

---

does the reported issue exist with the latest version of mujoco warp?

thanks!

jc-bao, 2025-07-04:
Hi @thowell , thanks for the quick fix! I pull the latest code and the issue persists. Could you run my control sequence and see if you can replicate that?

jc-bao, 2025-07-05:
Oh now it rendered correctly. I think I also need to uncomment the line below it. Thanks!

closed 2025-07-05 by jc-bao