google-deepmind/mujoco_warp #528: Hfield ncon balloons and should be culled further

btaba, 2025-07-22:
Currently prim<>hfield collisions are culled to mjMAXCONPAIR (50). In MJX, we culled to [4 contacts](https://github.com/google-deepmind/mujoco/blob/4f43a7d08e4f1289f4d76a4c48b8f42aca5bb38b/mjx/mujoco/mjx/_src/collision_convex.py#L1196), with no loss of generality (e.g. results in MuJoCo Playground). The 50 contacts quickly balloons memory over a large batch dim, and thus the contacts for prim<>hfield should be culled (ideally to a max of 4 contacts).

thowell, 2025-07-23:
should we make `MJ_MAXCONPAIR`
https://github.com/google-deepmind/mujoco_warp/blob/94d6928ebc51412cc4b1e58bd0815a023c3d0cc0/mujoco_warp/_src/types.py#L25
an option eg, `Option.maxconpair` and default the option value to 4?

btaba, 2025-09-10:
The tl;dr about why y'all might want to fix this before release is:

* max-contacts for a single geom <> hfield is set to 50
* these max numbers eventually get multiplied by num_envs O(10^3)
* this large constraint footprint for batch envs should be mitigated at some point
* we have some logic [here](https://github.com/google-deepmind/mujoco/blob/4b18009f47559a23e2c90b662769e0118b77c61e/mjx/mujoco/mjx/_src/collision_convex.py#L1202C18-L1207)  that fixes this, that could be replicated here

closed 2025-11-17 by thowell