google-deepmind/mujoco_warp #529: Bug: Mismatch between mjc and mjw rollouts when setting `kp` field in position actuators

alberthli, 2025-07-22:
### Description
This issue is related to #456. When setting non-default `kp` values in position actuators, the rollout error quickly grows, leading mjw to diverge from mjc.

### Environment Installation
Install a fresh environment:
```bash
conda create -n test_mjw_rollouts python=3.13
conda activate test_mjw_rollouts

pip install "mujoco>=3.3.3.dev" --pre -f https://py.mujoco.org/
pip install "warp-lang>=1.8.0.dev" --pre -f https://pypi.nvidia.com/warp-lang/
pip install git+https://github.com/google-deepmind/mujoco_warp.git
pip install packaging
```
Manually set `MJ_CCD_ITERATIONS = 32` in `io.py` in my site packages:
https://github.com/google-deepmind/mujoco_warp/blob/14fd6d39bef2122c2a5b7e82a0e723ffa81e09b0/mujoco_warp/_src/io.py#L28

Here's the output of `pip list`:
```
Package             Version
------------------- ------------------
absl-py             2.3.1
etils               1.13.0
fsspec              2025.7.0
glfw                2.9.0
importlib_resources 6.5.2
mujoco              3.3.5.dev785536018
mujoco-warp         0.0.1
numpy               2.3.1
packaging           25.0
pip                 25.1
PyOpenGL            3.1.9
setuptools          78.1.1
typing_extensions   4.14.1
warp-lang           1.9.0.dev20250722
wheel               0.45.1
zipp                3.23.0
```

### Test Data
I am using the following XML (slightly more information compared to #456):
```xml
<mujoco model="cylinder_push">
  <option timestep="0.02" />

  <asset>
    <texture name="blue_grid" type="2d" builtin="checker" rgb1=".02 .14 .44" rgb2=".27 .55 1" width="300" height="300" mark="edge" markrgb="1 1 1"/>
    <material name="blue_grid" texture="blue_grid" texrepeat="1 1" texuniform="true" reflectance=".2"/>
  </asset>

  <default>
    <default class="slider">
      <!-- THIS FAILS-->
      <position kp="10" ctrlrange="-10 10" forcerange="-1000 1000"/>
      <!-- THIS WORKS -->
      <!-- <position ctrlrange="-10 10" forcerange="-1000 1000"/> -->
    </default>
  </default>

  <worldbody>
    <body>
      <geom mass="0" name="floor" pos="0 0 -0.25" condim="3" size="10.0 10.0 0.10" rgba="0 1 1 1" type="box" material="blue_grid"/>
    </body>

    <body name="pusher" pos="0 0 0">
      <joint name="slider_x" type="slide" axis="1 0 0" />
      <joint name="slider_y" type="slide" axis="0 1 0" />
      <geom name="pusher" type="cylinder" size="0.25 0.1" mass="1" rgba=".9 .5 .5 1" friction="0.1"/>
      <site pos="0 0 0.15" name="pusher_site"/>
    </body>
[... 202 lines cut ...]
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                         strict=strict)
                         ^^^^^^^^^^^^^^
  File "/home/albert/miniconda3/envs/test_mjw_rollouts/lib/python3.13/site-packages/numpy/testing/_private/utils.py", line 926, in assert_array_compare
    raise AssertionError(msg)
AssertionError: 
Not equal to tolerance rtol=0.001, atol=0.001
Warp step does not match MuJoCo step
Mismatched elements: 24 / 160 (15%)
Max absolute difference among violations: 0.1638256
Max relative difference among violations: 0.83026095
 ACTUAL: array([[[ 0.378567,  0.952864,  0.733156,  0.615982,  0.201334,
          0.107466,  0.058084,  0.866176],
        [ 0.375969,  0.96112 ,  0.73951 ,  0.627267, -0.129893,...
 DESIRED: array([[[ 0.378567,  0.952864,  0.733156,  0.615982,  0.201334,
          0.107466,  0.058084,  0.866176],
        [ 0.375961,  0.961097,  0.739518,  0.627289, -0.130288,...
```
When I remove it, the tests pass:
```
MJWarp rollout took 0.0050s
Maximum absolute error between mjc and mjw states: 0.0001756089594620036
Maximum absolute error between mjc and mjw sensors: 2.8002066082555288e-05
Test passed!
```

thowell, 2025-07-25:
@alberthli thank you for opening an issue

not sure that this issue (#529) or issue #530 are bugs. the noted differences in rollouts may be due to differences in the floating point representations, float vs double, utilized by mujoco warp and mujoco, respectively. 

contact events might be particularly sensitive to such numerical differences.

with small timesteps (~0.001 seconds), a higher order integrator (RK4), and no active constraints (contacts, limits, etc), we should expect rollouts produced by mujoco warp and mujoco to match reasonably well.

```
import mujoco
import mujoco_warp as mjwarp
import numpy as np
import warp as wp

mjm = mujoco.MjModel.from_xml_path('cylinder_push.xml')
mjm.opt.timestep = 0.001
mjm.opt.integrator = mujoco.mjtIntegrator.mjINT_RK4
mjm.opt.disableflags |= mujoco.mjtDisableBit.mjDSBL_CONTACT

mjd = mujoco.MjData(mjm)

m = mjwarp.put_model(mjm)
d = mjwarp.make_data(mjm, nworld=1, nconmax=128, njmax=128)

with wp.ScopedCapture() as capture:
  mjwarp.step(m, d)

qpos0 = np.array([0, 0, .51, 0])
qvel0 = 1.0 * np.random.rand(mjm.nv)

mjd.qpos = qpos0
mjd.qvel = qvel0

wp.copy(d.qpos, wp.array(np.expand_dims(qpos0, axis=0), dtype=float))
wp.copy(d.qvel, wp.array(np.expand_dims(qvel0, axis=0), dtype=float))

for t in range(1024):
  if t % 64 == 0:
    print(f'{t}')
  
  action = np.random.rand(mjm.nu)

  # mujoco
  mjd.ctrl = action
  mujoco.mj_step(mjm, mjd)

  if mjd.ncon:
    print(f'mjd.ncon: {mjd.ncon}')

  # mujoco warp
  wp.copy(d.ctrl, wp.array(np.expand_dims(action, axis=0), dtype=float))

  wp.capture_launch(capture.graph)
  # mjwarp.step(m, d)

  if d.ncon.numpy()[0]:
    print(f'd.ncon: {d.ncon.numpy()[0]}')

  # compare
  error_qpos = np.linalg.norm(d.qpos.numpy()[0] - mjd.qpos)
  if error_qpos > 1.0e-2:
    print(f'qpos error: {error_qpos}\nmjd.qpos: {mjd.qpos}\nd.qpos: {d.qpos.numpy()[0]}')
    break

  error_qvel = np.linalg.norm(d.qvel.numpy()[0] - mjd.qvel)
  if error_qvel > 1.0e-2:
    print(f'qvel error: {error_qvel}\nmjd.qvel: {mjd.qvel}\nd.qvel: {d.qvel.numpy()[0]}')
    break
```

alberthli, 2025-07-25:
I can repro the results - this makes sense to me.