google-deepmind/mujoco_warp #550: Investigate performance of the n_humanoid scene.

Kenny-Vilella, 2025-07-28:
Following benchmark time given by #545, it looks like the scene with three humanoids is almost 10 times slower than with only one humanoid.
This scaling is not reasonable and we should understand what is happening.

Kenny-Vilella, 2025-07-28:
I check the nsight trace and it is correct.
48% of time spent in the blocked cholesky and 27.3% in JTDAJ.

JTDAJ is also slower than what it should be.

Kenny-Vilella, 2025-07-29:
For the blocked Cholesky, both nsight systems and nsight compute show wide variations of compute time during a single run, almost a factor 100.
From Nsight compute, I can see in slow kernels large cache miss, almost 200MB of transfer L2 cache <-> device memory and 600MB L1 cache <-> L2 cache.

Kenny-Vilella, 2025-12-11:
My impression is that perf is not yet as good as what it should be.
However, the introduction of island would probably solve some of the issue we are seeing with this benchmark.
So it's maybe not worth to spend weeks optimizing this type of case.

thowell, 2026-09-09:
islands + sleeping will not affect this scene since the humanoids are actuated and actuated trees do not currently sleep. additionally, this issue is a bit out of date now.

closed 2026-09-09 by thowell