google-deepmind/mujoco_warp #640: Unstable simulation with specific mesh asset

jc-bao, 2025-08-21:
# The issue

Simulations involving certain mesh assets are unstable. For example, the object in the provided XML file jumps continuously instead of resting on the ground.

# Reproduce

Load the following scene: https://github.com/jc-bao/mjwarp_test/blob/978e6e623c6d4ab83b4169de6bdc4db9adc7a8a8/assets/scene_coke.xml#L1

Run the command: `mjwarp-viewer scene_coke.xml --njmax=86`

# Expected Behavior

The object should settle and remain stable on the ground plane. At least for previous mjwap and mujoco cpu, it works fine. 

# Actual Behavior

The object is unstable and jumps off the ground.

https://github.com/user-attachments/assets/086d253d-db9f-4414-b836-b5736dd69e81

# Environment

mjwarp: baeb10e73237df13802217d310788e1f02dcf92d

mujoco: 3.3.6.dev797870883

warp-lang: 1.9.0.dev20250820

# Question

I have been unable to find a stable version by rolling back. Could you advise on a specific version combination that is known to be stable or suggest a potential fix?

Thank you!

thowell, 2025-08-26:
@jc-bao 

the simulation seems ok with
mujoco warp: 3f1197edd97a96a21384fcb418aa566306a22654
mujoco: 3.3.6.dev798987338
warp: 1.9.0.dev20250825

https://github.com/user-attachments/assets/e5d1442b-0ed1-4202-8e97-dfc1783de05d

please try the latest versions of mujoco warp + mujoco + warp and clear the warp cache

```
rm -r ~/.cache/warp
```

jc-bao, 2025-08-26:
Thanks @thowell ! I forgot to clear the cache so actually @btaba 's fix works now. I can also confirm the capsule <-> mesh collision issue has also gone thanks to @kbayes . 

Appreciate your quite response!

closed 2025-08-26 by jc-bao