google-deepmind/mujoco_warp #699: Capsule geoms on fingers penetrate grasped mesh object.

jamesheald, 2025-09-09:
Love the warp project! Thanks for all your great work.

I am applying it to a model in which a muscle-driven hand manipulates a mesh-based object.

I am noticing that the fingers can penetrate the object, allowing the hand to 'grasp' the object via finger penetration (I've attached a video showing this). This doesn't happen in mujoco.

https://github.com/user-attachments/assets/e2e28b06-2378-4127-bd0c-3ef51a8e1bf3

Capsule geoms are placed on the fingers for collisions. The general class of these geoms is:

<default class="myohand_coll">
    <geom type="capsule" contype="1" conaffinity="0" condim="3" margin="0.001" material="MatSkin"/>
</default>

My default mujoco model settings that I am also using in warp are:

integrator: Euler
solver: Netwon
timestep: 0.004
iterations: 100
ls_iterations: 100

The finger penetration behavior is observed if I set iterations and ls_iterations to much smaller values (e.g. 3-8), or if I set all geom margins to 0. It is observed with nconmax set to a large per-environment value (e.g. 500). 

Do you have any thoughts on what could be going on/how I might resolve this?

Thanks in advance.

jamesheald, 2025-09-09:
Amazing.

Here is the model (with assets): [model.zip](https://github.com/user-attachments/files/22234717/model.zip)

Thank you!

kbayes, 2025-09-15:
Hi @jamesheald, I was ill the end of last week. You can patch #713 in and see if this fixes your issue.