google-deepmind/mujoco_warp #734: [Question] Best practices for domain randomization on `pair_margin`

jc-bao, 2025-09-28:
Hi team,

I'm working on implementing domain randomization for contact parameters, specifically `pair_margin`, using `mujoco-warp`.

I've noticed that `mjModel.pair_margin` is initialized as a batched array based on `mjwarp._src.io.MAX_WORLDS`. This creates a surrogate memory mapping that seems to prevent overriding the margin for individual environments.

### What I've tried

My initial approach was to create a warp array with my desired randomized margin values and assign it directly to the model's `pair_margin`. However, this fails with a shape mismatch error because my number of environments (`num_samples`) doesn't match `MAX_WORLDS`.

```python
# Fails due to shape mismatch
num_samples = 1024
dev = "cuda"

default_model_wp = mjwarp.put_model(model_cpu)

# Create an override array with dimensions (num_samples, model_cpu.npair)
pair_margin_override_np = np.random.uniform(low=0.001, high=0.02, size=(num_samples, model_cpu.npair)).astype(np.float32)
pair_margin_override_wp = wp.from_numpy(pair_margin_override_np, dtype=wp.float32, device=dev)

# This assignment succeed but I cannot use it later
default_model_wp.pair_margin = pair_margin_override_wp
```

-----

### Current Workaround

My current solution is to manually modify the `MAX_WORLDS` constant to match my number of environments before initializing the simulation. This works, but it feels like a hack.

```python
# This works, but feels like an internal modification
import mujoco.warp as mjwarp

# NOTE: Hacky solution to align MAX_WORLDS with the number of envs
mjwarp._src.io.MAX_WORLDS = 1024 

# ... subsequent initialization code ...
```

-----

### Question

Is there a recommended or more elegant way to set `pair_margin` on a per-environment basis for domain randomization?

Thanks for the help\!

thowell, 2025-09-29:
@jc-bao please try the following code

```
import mujoco
import mujoco_warp as mjw
import warp as wp
import numpy as np

mjm = mujoco.MjModel.from_xml_string(
    """
  <mujoco>
    <worldbody>
      <geom name="plane" type="plane" size="10 10 .001"/>
      <body>
        <geom name="sphere" type="sphere" size=".1"/>
        <freejoint/>
      </body>
    </worldbody>
    <contact>
      <pair geom1="plane" geom2="sphere" margin=".123"/>
    </contact>
    <keyframe>
      <key qpos="0 0 .09 1 0 0 0"/>
    </keyframe>
  </mujoco>
  """
)

mjd = mujoco.MjData(mjm)
mujoco.mj_resetDataKeyframe(mjm, mjd, 0)

m = mjw.put_model(mjm)
d = mjw.put_data(mjm, mjd, nworld=2, nconmax=2, njmax=4)

mjw.step(m, d)
ncon = d.ncon.numpy()[0]
print(f"margin: {d.contact.includemargin.numpy()[:ncon]}")  # margin: [0.123 0.123]

pair_margin = wp.array(np.array([[0.25], [0.5]]), dtype=float)
m.pair_margin = pair_margin

mjw.step(m, d)
ncon = d.ncon.numpy()[0]
print(f"margin: {d.contact.includemargin.numpy()[:ncon]}")  # margin: [0.25 0.5 ]
```

thowell, 2025-10-05:
something like the following should work

```
m.pair_margin = wp.zeros((nworld, 1), dtype=float)  # update the field shape

with wp.ScopedCapture() as capture:
  mjw.step(m, d)

pair_margin = wp.array([[0.25], [0.5]]), dtype=float)
wp.copy(m.pair_margin, pair_margin)  # update the field elements in place

wp.capture_launch(capture.graph)
```

the field should be updated with the new shape prior to the graph capture. after the graph is captured, the field elements should be updated in place.

jc-bao, 2025-11-30:
Thank you so much Taylor! This is exactly what I needed.