google-deepmind/mujoco_warp #736: Heightfield terrain is very slow to compile and has very weird contact behavior

kevinzakka, 2025-09-29:
Attaching a big terrain composed of a mixture of sub-terrains. Some of the sub-terrains are heightfields. These heightfields make the compilation take forever (~5min) and the quadruped sinks when walking on it.

[rough_terrain.xml](https://github.com/user-attachments/files/22587706/rough_terrain.xml)

surajvh-ga, 2025-10-31:
There seems to be an indexing issue in the height field collision implementation. I wish I could take credit for this but this was pointed out by ChatGPT

https://github.com/google-deepmind/mujoco_warp/blob/main/mujoco_warp/_src/collision_convex.py#L463C7-L465

The indexing probably needs to change from
`nrow = hfield_nrow[g1]`
`ncol = hfield_ncol[g1]`
`size = hfield_size[g1]`
to 
`nrow = hfield_nrow[geom1_dataid]`
`ncol = hfield_ncol[geom1_dataid]`
`size = hfield_size[geom1_dataid]`

Tagging the issue #815 here as well

This fixes the sinking of the robot into the ground that @kevinzakka is mentioning

thowell, 2025-10-31:
@surajvh-ga will you please submit a pr with these changes? thanks!

surajvh-ga, 2025-10-31:
@thowell I've raised a pull request

Abban-Fahim, 2025-11-02:
@kevinzakka I came across the same issue you described when trying to use the rough-terrain environment in playground. In my case it just crashes (killed by OOM, no python backtracre) when using `impl=warp`, while the flat-terrain works perfectly fine. This is despite my XML having way fewer bodies in the terrain. I suspect that this is because of my limited RAM (<= 16GB), and my process gets killed immediately. Is there any progress in speeding up the compilation step / reducing the memory footprint? I'm using the latest version of this repo, e605c406363805d949b75d50dd5ffb3d79bc3f70.

thowell, 2026-01-06:
@kevinzakka is this resolved?

kevinzakka, 2026-01-06:
So far resolved, will re-open if I encounter further issues.